designer2k2/EMUcan #3: Wrong IgnAngle handling

pawliczka, 2022-07-21:
Signed data needs to be cast to signed before assignment because it causes wrong negative value handling:
For example ignition angle -1 is stored as 125.

Can data use unsigned 8 bit
__u8    data[CAN_MAX_DLEN] __attribute__((aligned(8)));

To solve the problem change:
`emu_data.IgnAngle = msg->data[0] / 2;`
to
`emu_data.IgnAngle = int8_t(msg->data[0]) / 2.0;`

https://github.com/designer2k2/EMUcan/blob/113c9ace30fd7afe5706ab98213da8e4bec02d4e/src/EMUcan.cpp#L129
This problem also applies to other signed values.

designer2k2, 2022-07-21:
Hello, thanks for reporting that! And also thanks for providing a solution!
Looking at a logfile its so clear to see that the sign on the IgnAngle is wrong, i have no idea how i have missed that.
Also prooves that ive never startet the engine below freezing as IAT and CLT is also signed...
Will provide a fix on the weekend, need to check this in a car.

designer2k2, 2022-07-21:
bugfix in work: 76359218ed530fd22712281c93a3ba7c425a5ca9  (untested currently)

pawliczka, 2022-07-22:
You have to divide it by 2.0, not 2. Because for example when the ignition angle is 7 and you divide it by 2 the result is 3, not 3.5
https://onlinegdb.com/XdHhvB1uf

designer2k2, 2022-07-22:
yes that’s also wrong.  I went ahead and replaced it with multiplication as its faster than division on AVR cores: 9546952349870e8017b0800fc4ab08463a659222

closed 2022-07-23 by designer2k2

designer2k2, 2022-07-23:
tested ok in a car and released as new version

PR #4 (2022-07-23): Signed int bugfix
this fixes #3