designer2k2/EMUcan #14: Increase update speed?

coultl6556, 2023-05-07:
Got things working with my ECUMaster Black. So cool. Running an Arduino Nano Every and a MCP2515 breakout board. I'd like to use the MAP data to run a boost gauge (stepper motor and run by the Arduino) but the update speed seems too slow for this to work. It seems like it updates at something like 1hz. Is there a limitation that I'm hitting or should it be faster? I think I'd need it to pull values at about 50hz minimum to get a gauge to work decent.  Any tip on how to get a faster update on the EMU values over CAN? Thank you!

Video of my custom boost gauge here: https://youtu.be/ZjTRC6D3Vsw

designer2k2, 2023-05-08:
Hello,
that´s a nice integration 👍 

The Black is sending the MAP with 20Hz, see this File on a PC with the Software for the Black installed:
file:///C:/Program%20Files%20(x86)/Ecumaster/EMU%20Black/Help/EN/emuCANStream.html

This 20Hz are enough to give a smooth view on moving display´s.

EMUcan is capable to receive this 20Hz easily, but getting the full speed on a Nano Every needs some care taken in the software.
Even on a strong Teensy 4 (600MHz clock speed) this needs some considerations.

A good example is the Arduino Blink without Delay: https://docs.arduino.cc/built-in-examples/digital/BlinkWithoutDelay

This practise of running a loop at maximum speed to capture every CAN frame and then taking actions on given intervalls works well on Arduinos.  The same way i build the examples provided with EMUcan.

In the video was also a small display, how fast is this update? Some displays take quite a while to send the data to. 
And how is the stepper driven?  Is the code blocked while the needle is moving?

designer2k2, 2023-05-08:
Yes lets keep this open here.

(i modified your comment to have the code formatted)

The programm looks ok, one thing is that you update the display on very loop.  That will slow down the loop a lot, try to get the loop running much faster and update the display at a given intervall.

Place the display update in the "Serial out every second" but let it update like every 250ms.

coultl6556, 2023-05-08:
Well, that totally worked...perfectly. Gauge works great, super responsive. Screen is updating even faster now. Something about updating the screen on every loop was slowing everything way down. 

One note for others if they build something similar. I used the TPS value to test the gauge that way I can get instant feedback when I push the pedal with the car off. 

Updated code here in case it's helpful to anyone:

`// EMUCan Lib Simple Test Example

// Example to be run on Arduino (Nano) with MCP2515
// Configure the EMU Black to send the CAN Stream at 500KBPS

// This MCP2515 Lib is used:
// https://github.com/autowp/arduino-mcp2515

// https://www.designer2k2.at
// Stephan Martin 27.12.2022

#include "EMUcan.h"
// EMU initialized with base ID 600:
EMUcan emucan(0x600);

int iat = 0;

//screen code
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 32 // OLED display height, in pixels
// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
// The pins for I2C are defined by the Wire-library. 
// On an arduino UNO:       A4(SDA), A5(SCL)
// On an arduino MEGA 2560: 20(SDA), 21(SCL)
// On an arduino LEONARDO:   2(SDA),  3(SCL), ...
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)
#define SCREEN_ADDRESS 0x3C ///< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// needle code
#include "SwitecX25.h"

#include <mcp2515.h>
struct can_frame canMsg;
// MCP2515 with chip select (CS) pin 10:
MCP2515 mcp2515(10);

// Code for stepper needle
// 315 degrees of range = 315x3 steps = 945 steps - changed to 630 steps
// declare motor1 with 945 steps on pins 4-7
SwitecX25 motor1(630,3,4,7,8);
// Boost in kPa
int boost = 0;
int needlepos = 0;

unsigned long previousMillis = 0;
const long interval = 250;

static const unsigned char PROGMEM quattro_glcd_bmp[] =
{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xE0, 0x03, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xE0, 0x03, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00,
[... 120 lines cut ...]
    //if (emucan.EMUcan_Status() == EMUcan_RECEIVED_WITHIN_LAST_SECOND) {
      //Serial.print(emucan.emu_data.RPM);
      //Serial.print(";");
      //Serial.print(emucan.emu_data.TPS);
      //Serial.print(";");
      //Serial.print(emucan.emu_data.IAT);
      //Serial.print(";");
      //Serial.print(emucan.emu_data.MAP);
      //Serial.print(";");
      //Serial.println(emucan.emu_data.pulseWidth);
   // } else {
   //   Serial.println("No communication from EMU");
    // }
   // if (emucan.emu_data.flags1 & emucan.F_IDLE) {
     // Serial.println("Engine Idle active");
   // }
   // if (emucan.decodeCel()) {
   //   Serial.println("WARNING Engine CEL active");
  //  }
 // }
  
  //display.clearDisplay();

}
`

designer2k2, 2023-05-09:
That looks great in the video! 💯 

One further tweak could be to take **setTextColor** and  **setTextSize** out of the loop. If its never changed it can go in setup. And therefore speed up the loop further. 
Just would need to be tested as im not shure if the Adafruit_SSD1306 will remember this or the setting is needed every time.

Small note on the code, are you using the Arduino IDE? For readability try:
Selecting Edit > Auto Format or pressing CTRL + T on Windows/Linux or CMD + T on MacOS when writing a sketch in the Arduino IDE 2 will automatically format the sketch
This will format, or better call it indent the code to make it "nicer" to read.

coultl6556, 2023-05-09:
Will do. Thank you!

closed 2023-05-10 by coultl6556