designer2k2/EMUcan #20: Possible to run EMUcan on a Lilygo T-Panel S3?

TobyLies, 2024-08-06:
I'm looking for a simple integrated ESP32 solution and I found this Lilygo T-Panel S3: https://www.lilygo.cc/en-ca/products/t-panel-s3?variant=44273605509301

It's available in CANFD. Would this work with the ECUMaster and EMUcan?

My hope is that I could use this T-Panel as a complete gauge solution.

designer2k2, 2024-08-07:
Hello,
oh thats a cool device!

Basically every device works with EMUcan, as you just pass the CAN-Frame into it. 

The ESP32-S3 has the TWAI (CAN) interface, so the ESP32 example from here should work: https://github.com/designer2k2/EMUcan/blob/main/examples/EMUcan_ESP32/EMUcan_ESP32.ino

Here the info on the TWAI on ESP32-S3: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/twai.html

Only the CANFD part is confusing, as the ESP32 directly cannot speak CANFD, so how does the T-Panel make this work?
In the documentation i cannot find info on the CANFD part, no example, no schematics: https://github.com/Xinyuan-LilyGO/T-Display-S3

Do you have there anything more detailed?

TobyLies, 2024-08-07:
From my limited knowledge, I don't think that TWAI works with CANFD? From what I gather, it will interpret FD frames as errors.

There is a schematic here: https://github.com/Xinyuan-LilyGO/T-Panel/tree/arduino-esp32-libs_V2.0.14/project Page 3 shows the CAN connections.

Note I'm looking at the T-Panel, not the T-Display.

If using the CANFD version isn't possible, could I use the RS485 version of the T-Panel with an external can board like this?: https://copperhilltech.com/mcp2515-can-bus-breakout-board-with-spi-interface-2-pack/

designer2k2, 2024-08-07:
yes the ESP32 natively does not support CANFD.  do you need that? for the EMUBlack its not needed.

Ah i was on the wrong page, the T-Panel schematic is showing that the CAN is wired over a transceiver into the ESP32. This means it will work with the ESP32 example from here (once you adapt the used Pins). But it will not support the claimed CANFD.

TobyLies, 2024-08-07:
No I don't need CANFD, my only knowledge of it is from the product page.

I think I'll just order it and give it a shot. Thanks for your help, and thanks for making EMUcan.

designer2k2, 2024-08-07:
for the used pins, there are some definitions for the T-Panel given:
https://github.com/Xinyuan-LilyGO/T-Panel/blob/803c9e70e6b17b0b6a557091d97df4720e734d0f/libraries/Mylibrary/pin_config.h#L42
``#define CAN_TX 16``
``#define CAN_RX 15``

so change it in the ESP32 example to pins 16/15: https://github.com/designer2k2/EMUcan/blob/main/examples/EMUcan_ESP32/EMUcan_ESP32.ino  

This should work right away, please keep me updated!

TobyLies, 2024-08-21:
Got the T-Panel, and the sketch uploads to it fine with the pin change. My car is parked elsewhere so I can't connect the CAN at the moment.

closed 2025-03-03 by designer2k2

sskramgarhia, 2025-03-12:
> Got the T-Panel, and the sketch uploads to it fine with the pin change. My car is parked elsewhere so I can't connect the CAN at the moment.

Hi Toby,

I am working on the similar project, do you think you can share what version you went with CANFD or RS485.
Also if you have any suggestion to make the code work.

Thanks