designer2k2/EMUcan #31: Very slow values refresh rate, ESP 32 with SN65HVD230.

techno2235, 2025-02-14:
Hi, i have problem with reading EMU black CAN values fast enough, i have tested this off car ECU powerd on bench and CAN bus wired only, MAP value is updating like every 2/3 seconds in serial monitor or on OLED display using SPI, testing code used was EMUcan_ESP32 example without any modifications, i have also modified the code to used OLED display and got rid off serial monitor print part, didnt change anything. Changing serial print time wasnt changing anything, except printing same value more times. When i used display i had simulation (not releated to can only display testing) of oil temp running at the same time and it was updating really fast something around few ms (as fast as display could), and MAP still updated with huge delay. So i would assume that my loop is running fast enough, but doesnt read CAN bus fast enough? Im lost and have no idea what to do.
Tested with MAP by using syringe connectec to Emu
As for connections, terminnating Can bus in EMU, using 120 ohm resistor, both or none didnt change a thing in delay time, 

Thats my first project using electronics and programming, .There can be many amateur mistakes that i dont see :).

Thanks in advance for help.

designer2k2, 2025-02-14:
Hello,
to crosscheck with your setup, if you look into the EMU Logfile with a PC connected to the EMU is there the MAP Signal fast? There are things like the "Map filter" that can slow down the Signal a lot. 

The EMUcan_ESP32.ino example prints every half second (500ms) to the serial port. Is there for example the TPS working fast?

techno2235, 2025-02-15:
So, here is video how it looks like https://drive.google.com/file/d/1zt3Adh1vRtvj8ICZ8joea2VW7H9SXSJ9/view
Emu is out of car, and im not able to wire TPS right now, I will test later how with something else wired.
Changing print time to 100ms only makes same value printed more often, I honestly dont see what makes code in Issue #14 work better, only that part is missing   

uint32_t alerts_triggered;
  twai_read_alerts(&alerts_triggered, pdMS_TO_TICKS(1));
  if (alerts_triggered & TWAI_ALERT_RX_DATA)

but deleting that doesnt change anything.

designer2k2, 2025-02-15:
yes the video looks wrong!

to see if there is something on the ESP32 side i extended the example with additional debug output. See here: https://gist.github.com/designer2k2/593196a4964f4187830343b95217acdb

This prints additional lines in case of errors

techno2235, 2025-02-16:
Here is what serial looks like right after resetting with button on esp, it goes on and on

15:27:40.926 -> Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
15:27:40.926 -> Bus error count: 26
15:27:40.926 -> Alert: The RX queue is full causing a received frame to be lost.
15:27:40.926 -> RX buffered: 0	RX missed: 139	RX overrun 0
15:27:40.926 -> Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
15:27:40.959 -> Bus error count: 34
15:27:40.959 -> Alert: The RX queue is full causing a received frame to be lost.
15:27:40.959 -> RX buffered: 0	RX missed: 201	RX overrun 0
15:27:40.959 -> Alert: The RX queue is full causing a received frame to be lost.
15:27:40.959 -> RX buffered: 0	RX missed: 271	RX overrun 0
15:27:40.959 -> Alert: TWAI controller has become error passive.
15:27:40.959 -> Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
15:27:40.959 -> Bus error count: 48

here is some lines copied after some time 

15:29:10.912 -> RX buffered: 0	RX missed: 260903	RX overrun 0
15:29:10.912 -> Alert: The RX queue is full causing a received frame to be lost.
15:29:10.912 -> RX buffered: 0	RX missed: 260935	RX overrun 0
15:29:10.912 -> Alert: TWAI controller has become error passive.
15:29:10.912 -> Alert: A (Bit, Stuff, CRC, Form, ACK) error has occurred on the bus.
15:29:10.912 -> Bus error count: 30338
15:29:10.912 -> Alert: The RX queue is full causing a received frame to be lost.
15:29:10.945 -> RX buffered: 0	RX missed: 260952	RX overrun 0
15:29:10.945 -> Alert: The RX queue is full causing a received frame to be lost.
15:29:10.945 -> RX buffered: 0	RX missed: 261022	RX overrun 0
15:29:10.945 -> Alert: The RX queue is full causing a received frame to be lost.
15:29:10.945 -> RX buffered: 0	RX missed: 261055	RX overrun 0
15:29:10.945 -> Alert: The RX queue is full causing a received frame to be lost.
15:29:10.978 -> RX buffered: 0	RX missed: 261088	RX overrun 0

designer2k2, 2025-02-16:
oh thats bad, but its part my fault as i copied my local example where im using the listen only mode. I assume you only have Emu Black and the ESP32? That requires the normal mode.

I updated the example from above, please run this again: https://gist.github.com/designer2k2/593196a4964f4187830343b95217acdb#file-esp32_advanced-ino-L34

techno2235, 2025-02-16:
I have connected it together on my  desk, only EMU, ESP32, CAN transciver SN65HVD230, for getting it working, before i finish with car and istall it there.

Right now it works perfect, when looking at serial compared to Emu log, will adapt that code to work with display, hope it will work as well as with serial. No error messages in serial only values and flags being printed.

As for code changes the thing that made it work was TWAI_MODE switched to NORMAL from what i see.
And added part for queue lenght changing it to 10? How that value affects CAN reading? Response time? Amount of messages processed at once?

Quick question does your library has functions to send messages to EMU, didnt see any mentions in readme file or do i need to setup custom message using twai library? (sending switch state or analog input values) Maybe in future if i lack EMU inputs i would work on that, so not anything important.

Thank you so much for help and very fast response, have a nice day :)

designer2k2, 2025-02-16:
Great! Good to hear you have it working now!  I will push a update soon with the fixed ESP32 example.

Yes the NORMAL mode has fixed it, the queue lenght can be taken out for this example.  
The queue length is needed in case you run some intensive other work (like updating a large display). Then it helps to ensure no frame gets dropped.  If you keep the alerts in the code you can see if its needed in future.

Sending is supported with an example, it does not need a lib as its so easy: https://github.com/designer2k2/EMUcan/blob/main/examples/EMUcan_MCP2515_send/EMUcan_MCP2515_send.ino
This needs some adoption for the ESP32, take a look at the transmit example from ESP32: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/TWAI/TWAItransmit/TWAItransmit.ino

designer2k2, 2025-02-16:
this is fixed in 2.0.3: https://github.com/designer2k2/EMUcan/releases/tag/2.0.3

techno2235, 2025-02-17:
Thanks for help again, have a nice day.

closed 2025-02-17 by techno2235