designer2k2/EMUcan #37: EMUcan, stream button state to EMU

error105, 2025-03-30:
Hi, can somebody explain how can i use digital button on my ESP32 connected via CAN to EMUblack and use that status in EMU? Eg, activate cruise or change maps?

error105, 2025-03-30:
I mean, how to use stream in EMUblack - eh, if im push button in ESP i want activate Cruise :)

designer2k2, 2025-03-31:
Hello,
did you take a look at the send example? https://github.com/designer2k2/EMUcan/tree/main/examples/EMUcan_MCP2515_send

it streams a analog value to the EMUblack, but the idea would be the same for a digital value

error105, 2025-04-07:
Sorry for delays :)
I canttest itrightnow, but im think that we need also add heartbreath++ and that frames should sends every 20ms like documentation say, am i right?

void Send_CAN_Switch_States() {
  if (emucan.EMUcan_Status() == EMUcan_RECEIVED_WITHIN_LAST_SECOND) {
    byte CAN_Switch_State = 0;

    bool CAN_Switch_1 = digitalRead(SWITCH_1_PIN);
    bool CAN_Switch_2 = digitalRead(SWITCH_1_PIN);

    bitWrite(CAN_Switch_State, 0, CAN_Switch_1);
    bitWrite(CAN_Switch_State, 1, CAN_Switch_2);

    // Configure message to transmit
    twai_message_t message;
    message.identifier = 0x0F7;
    message.data_length_code = 1;
    message.data[0] = CAN_Switch_State;
    message.data[7] = heartbeat++;
}

designer2k2, 2025-04-07:
You mean the CAN Switch Board V3? https://www.ecumaster.com/files/devices/switchboard/switchboardManual.pdf

The heartbeat is not needed, it works also if you send it without that. 
Also the update intervall can be set as wished. Ideally you would even only send if the state of a switch has changed. Try to keep the load on the CAN Bus low.

If you want to include the hearbeat make shure that the DLC matches the byte count.