designer2k2/EMUcan #39: Cruise Flag

error105, 2025-05-03:
Hi,there is any chance that u will add Cruise control flag? Or can u explain how i can read with EMUcan if Cruise is active?

designer2k2, 2025-05-03:
Hello,
cruise control is not part of the default CAN stream, it has to be added with the user defined CAN stream.
Like so:
[image]
It will send every 500ms the status from the Cruise control active with message 555h.

Now in Code you need to check for this message and read it. 

based on this: https://github.com/designer2k2/EMUcan/blob/main/examples/EMUcan_ESP32/EMUcan_ESP32.ino

Replace from 70:
```c++
    while (twai_receive(&message, 0) == ESP_OK) {
      emucan.checkEMUcan(message.identifier, message.data_length_code, message.data);
    }
```

with:
```c++
    while (twai_receive(&message, 0) == ESP_OK) {
      emucan.checkEMUcan(message.identifier, message.data_length_code, message.data);
      if (message.identifier == 555h) {
            bool cruisecontrol;
            cruisecontrol = message.data[0] & (1 << 0);
            if (cruisecontrol ) {
              Serial.println("Cruise control active");
            }
      }
    }
```

This code is not tested! But the idea is ok :-)

designer2k2, 2025-05-04:
Hello,
yes using the virtual output is a good way! Your code is ok for this but could be even shorter:
```c++
CRUISE = emucan.emu_data.outflags1 & emucan.F_VPO3;
```

the last example would not work, you need to use the canMsg.data when checking for cruisecontrol. Also the bool should be defined outside or it is not visible for other functions.

see:
```c++
bool cruisecontrol;
void CANread(void *parameter) {
    for (;;) {
        while (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
            emucan.checkEMUcan(canMsg.can_id, canMsg.can_dlc, canMsg.data);
            if (canMsg.can_id == 0x555) {
                cruisecontrol = canMsg.data[0] & (1 << 0);
            }
        }
    vTaskDelay(pdMS_TO_TICKS(50));
  }
}
```